fluidd-core/fluidd #322: Allow basic auth in Moonraker url

RafaelSchridi, 2021-06-16:
## Is your feature request related to a problem? Please describe.
I've got my moonraker instance behind a basic HTTP auth, which you access by passing the username and password in the url like shown in the screenshot below.
Currently the form blocks because of the strict regex

## Describe the solution you'd like:
Allowing any characters within the url like this would be great
**https://\*:\*@example.com**

The fix is probably in this regex:
https://github.com/cadriel/fluidd/blob/7e8785d8c30afd80ce85d981a82f9992b525254c/src/components/common/AddInstanceDialog.vue#L94-L99
but I'm too much a dumbass at advanced regex to fix it myself

## Describe alternatives you've considered:
Tried logging in another tab to use the auth in fluidd but that doesn't work.

## Affected users and/or setups:
probably a very small minority who use reverse proxies

## Additional Context:
[image]
Here it is working in postman:
[image]

cadriel, 2021-06-16:
An incoming update adds JWT authentication, natively within Fluidd. I believe this would remove the need for you to have this?

RafaelSchridi, 2021-06-16:
JWT authentication would indeed remove my need for this, but there are some people who are swear by and only use basic auth. But if it's more than just a regex edit it's not worth the effort.

wlhlm, 2021-06-18:
I think if implementation mostly centers around fixing this regex, then I would still like to see this implemented. On the surface, basic auth seems less complex and harder to get wrong compared to JWT.

RafaelSchridi, 2021-06-18:
More options wouldn't hurt right?
Also the webcam URL currently supports basic auth, so why shouldn't the moonraker url also support it?

cadriel, 2021-06-20:
I've had some more thoughts on this, and done a little research. Passing credentials directly in the URL isn't great practice - especially plain text passwords, and some browsers refuse it, while others have it deprecated.

For those reasons I'd recommend not passing creds in the URL. We could look at defining basic auth as separate inputs to then pass in the request header as an option, but I'd prefer users simply enable auth in moonraker and use the built in mechanisms, or at least give that a try first - as its much more secure.

cadriel, 2021-06-29:
Any further thoughts on this now that Auth has been released?

RafaelSchridi, 2021-06-29:
I have switched over to JWT, so for me this is no longer necessary.

closed 2021-07-01 by cadriel