fluidd-core/fluidd #327: [Feature Request] Configurable [power] object g-code

alchemyEngine, 2021-06-21:
Hey Cadriel,

I brought this up in https://github.com/Arksine/moonraker/issues/181 already, and it doesn't seem like it'll be a good fit for Moonraker itself.

Basically, I use [power] to toggle the main PSU in my printer, therefore I need to park the toolhead and disable the stepper motors before powering down every time. I was wondering if you think a configurable `gcode power off` and maybe `gcode power on` setting for each [power] object might also be useful to other users?

PS. love fluidd, thanks a lot for your work here.
-alch3my

cadriel, 2021-06-22:
Hey - I've taken a look at this - and I *think* what you're asking for is the ability to swap the power toggle for a gcode macro of your choosing?

I.e., you can already accomplish what you want via a gcode macro - so it sounds like what you're really after is the power toggle in the hamburger menu to invoke your macro instead of firing the power API directly?

alchemyEngine, 2021-06-22:
Essentially that, yes. Ideally the toggle would still indicate the power state via the API though.

pingyofdoom, 2021-08-23:
I have a 64 bit OS, I think it has some GPIO compatibility problem, so the power button isn't coming up. Having the ability to enable a customizable power button that could run g-code would be really beneficial. But, G-code doesn't work while the printer is off? How does Fluidd control the Power plugin?

pedrolamas, 2022-02-18:
I believe this is already implemented in Moonraker by using a `[power]` with a `type: klipper_device`

Check [here](https://moonraker.readthedocs.io/en/latest/configuration/#klipper-device-configuration) for more information, specifically where it mentions "gcode_macro"

closed 2022-02-18 by pedrolamas