fluidd-core/fluidd #331: FR: Add information about the used filament / needed material weight in gram

TimoErdmann, 2021-06-22:
Fluidd already shows the needed Material in length (meters). it would be handy to visualize the needed Material in weight (gramm).  So if there is not enough  material on one spool one can do a spool change mid print and stay safe to have enough material on the new spool. 

I dry a full spool of material and respool the needed amount of material on an empty spool and put the other spool back in the drying cabin. so it keeps dry and i dont have to redry the remaining material again. But it happens that there is not enough material on the dry spool. So i need to change the material mid print but i am able to start the print right now. 

In the print files is the needed materiallength in meters given. To calculate the needed materialweigth there are two options: 
1. we need to know the density and the diameter of the filament. The diameter is given in the klipper config. but the density depends on the used material. All print files have the material in their names (this is a convention of the files created in the slicer). So it would be possible to get the materialtype per regular expression and look up the density which is stored in a json file.
2. Use the weight of the print. The slicer adds this information too to the filename. Analog to the materialtype the weight could be parsed and with the rule of three the used and needed weight could be calculated with the materiallength.

## Is your feature request related to a problem? Please describe.
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

## Describe the solution you'd like:
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

## Describe alternatives you've considered:
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

## Affected users and/or setups:
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

## Additional Context:
(Add any other context or screenshots about the feature request here.)

maruhe, 2021-07-03:
f.e. in Cura you can enter the density for each filament-type

cadriel, 2021-07-03:
Fluidd already displays the material weight via metadata in your gcode. You can enable this column by configuring your jobs panel to show it.

In order to have any other data it'd need to be supplied via moonraker.

What specifically would you want to display over and above the already provided weight?

closed 2021-07-08 by cadriel

TimoErdmann, 2021-12-10:
hey i have looked into it. I am able to show the needed Material weight now in Fluidd after adding it to the header of the print file. But i want to have more information about the current print job. It is regularly the case, that i have print jobs which need more material that is on the spools i have laying around. I have to change the spool of filament regularly and therefore i check the weight of the spool. 

To make this process easier it would be convenient to show the printed Material in length and weight! 
It could look like this:
[image]
 
As an alternative it would be nice to show the material weight which still has to be printed when hovering over those numbers.