fluidd-core/fluidd #332: [v1.15.0] Movable Neopixel Color Picker.

Mitsuma, 2021-06-22:
## Is your feature request related to a problem? Please describe.
When trying out the new Neopixel Color Picker I noticed that the window that opens is not movable.
This makes it a bit hard to see the changes if your layout has the camera/console above the color picker.

## Describe the solution you'd like:
Simply make the color picker movable, freely movable would be the best, like click and drag it around. 

## Affected users and/or setups:
Would help people who have their fan controls and camera a bit separate. Just a small change.

Ramalama2, 2021-09-10:
@cadriel is it possible to add an color picker for rgb leds? Where you control each pin (red/green/blue) with pwm from 0.00 till 1.00 🙈
This would be amazing. Lets say if the names in the config files, matches something you like 🙈

If not, thanks anyway cadriel, thanks for fluidd it's amazing already as it is ♥️

matmen, 2022-03-14:
@Nailig Sorry, but I can't reproduce this. Can you clear your browser cache and see if it still happens? For reference, this is the behavior on my end:
[image]

PR #577 (2022-03-16): feat(ui): make color picker movable
Adds a draggable title to the color picker card - now they can be moved anywhere you'd like ;)
[image]

Closes #332
closed 2022-03-16 by matmen