fluidd-core/fluidd #335: BME280

Afterschnitzel, 2021-06-23:
Hi. 
Is possible to show data from BME280 some where in temperature window (temperature, humidity, barometric pressure) without using macro in terminal?

Thanks in advance.

cadriel, 2021-06-24:
I'll take a look - its hard to implement this without having one tho.

Afterschnitzel, 2021-06-24:
Thank you. It is not expensive to afford one :-)

pedrolamas, 2022-02-18:
Related to #434, I assume we can retrieve at least the humidity using the same approach

samwiseg0, 2022-02-18:
> Related to #434, I assume we can retrieve at least the humidity using the same approach

From my understanding, The pressure will be present in the object using the BME280. In my example I do not have it as a value as my sensor does not support barometric pressure.

pedrolamas, 2022-02-18:
@samwiseg0 indeed, seems to be the case!

I've got a few BME280's somewhere, I just need to find them and will start some tests...

PR #538 (2022-03-19): feat: show pressure and humidity from sensors
Here's how it is currently showing the humidity output from a BME280 connected to a Pi Zero 2W running as MCU:

[image]

Fixes #335, fixes #397, fixes #434
closed 2022-03-19 by pedrolamas