fluidd-core/fluidd #344: needet disabled controls

maruhe, 2021-06-29:
first of all last release was great, meanwhile I use 90% fluidd (9% octoprint and 1% mainsail)

while sd-printing the controls to change the mesh view are disabled
I think all controls in the red box can / should be active whenever a mesh is loadet
! [graphic] (https://user-images.githubusercontent.com/49123648/123870231-5609b100-d932-11eb-846e-05514e56ffb1.png)

at the dashboard the control for speed and flow get disabled on change until klipper responds => this means when I enter a wrong value while my prepare_print-macro is running I need to open mainsail to fix these values before the printer starts printing

i hope this changes fit your mind and can be done by adding some # 's :-)

Thank you for creating fluid!

oh, and the option would be great @console-widget to show the text-field and the send-button above output-window

cadriel, 2021-06-30:
The mesh controls are fixed in the next release already.

Re: your second point, I'm not sure what you mean. Can you give me an example of an incorrect value?

maruhe, 2021-06-30:
[image]

here you see the printer is executing my prepare_print-macro since minutes (bed is heating)
i changed speed to 75% and now the control is disabled because the printer is not responding

incase 75% would be the wrong value I now would have to wait until the printer has started printing to change the speed again
sure I could do it by command..

cadriel, 2021-07-02:
I've taken another look at this - and, Fluidd is behaving as expected.

I.e., during a wait in Klipper (when temps are coming up) - if you send a speed change - it does not get applied until after the wait cycle is complete.

In your example above - yes, the control is disabled - because it gets no response. However - if you wait for the heating cycle to complete - you'll notice that the control eventually re-enables itself.

In mainsail's case, it does not disable the control - however, none of the speed commands actually apply either - and so I think the way Fluidd handles this is better because it gives the user feedback in that the command hasn't processed yet.

maruhe, 2021-07-03:
absolute, in general I see this point and perspective!

BabyStepping is the other way araound - in Fluidd I can press the buttons multiple so I can set the value I need
in mainsail this button get disabled until printer response wich can be quit dangerous

>if you wait for the heating cycle to complete - you'll notice that the control eventually re-enables itself.
for sure, I already noticed, just my heatup-and calibration script can need some time - this 400mm bed's are slow on higher temp..

and yess, it's great to see that a command is not yet executed
anyway sometimes I have to change this values multiple while printer is not responding in that momemt
may a compromise could be to let the edit enabled? so I see (usually I even know) the printer is not responding by the disabled slider/button and still could enter a value :-D

>and so I think the way Fluidd handles this is better
yess it does 👍

StuSerious, 2022-01-30:
@maruhe I think Cadriel addressed it properly and that this can be resolved, would you agree?

matmen, 2022-02-25:
@maruhe Any updates on this? Can this be closed?