fluidd-core/fluidd #347: Can't host on a subdirectory

Salamandar, 2021-07-02:
## Affected Version(s):
- fluidd version: v1.15.0

## Platform:
Octopi

## Reproduction Steps:
I tried to set up a reverse proxy with nginx to access fluidd with an url like : `https://print.myhost.com/ender_fluidd`.
It's impossible because resources are loaded from an absolute path (`/js/something.js`) and it's not relocatable to `/ender_fluidd/js/something.js`.

The idea was to get octoprint and fluidd aside, or multiple fluidd instances on the same port (https). It seems it's impossible ?

cadriel, 2021-07-02:
This is not a bug. Fluidd is a PWA and SPA and requires hosting on the root of a domain.

I'd instead suggest a reverse proxy to subdomain.

closed 2021-07-02 by cadriel

Salamandar, 2021-07-03:
I don't really understand the limitation you're talking about. This is based on Vue, right ? And Vue apps can be hosted on a subpath. I already saw multiple ones.
https://daten-und-bass.io/blog/serving-a-vue-cli-production-build-on-a-sub-path-with-nginx/

Sureiya, 2021-11-16:
@Salamandar He's got things reversed. You can't have a SPA navigate across subdomains without a refresh. You can have as many as you want on one though, if they're written to support it.