fluidd-core/fluidd #349: Removing built-in macros or having the ability to arrange them.

lar3ry, 2021-07-04:
Nice software!

## Is your feature request related to a problem? Please describe.

Interspersing of built-in macros with my own.

## Describe the solution you'd like:

I'd like to be able to rearrange, rename, or remove built-in macros (CANCEL_PRINT, PAUSE, RESUME)

## Describe alternatives you've considered:

Tried renaming CANCEL_PRINT to X_CANCEL_PRINT. This resulted in not being able to get to the home screen at all.

Tried removing the call to the built-in macro file. This allowed me to get to the home screen and use the printer, but at the cost of a big gaudy warning that eats vertical space.

## Affected users and/or setups:

Anyone who wants to group or rearrange their macros.

## Additional Context:
(Add any other context or screenshots about the feature request here.)

cadriel, 2021-07-04:
What's wrong with turning their visibility off? This is something you can already do...

lar3ry, 2021-07-04:
> What's wrong with turning their visibility off? This is something you can already do...

Nothing wrong with that at all! 
It's easy when you know about that. Thanks!

closed 2021-07-04 by lar3ry