fluidd-core/fluidd #350: Monaco Editor - Klipper config style inline comments and values not highlighted properly

MrAntza99, 2021-07-04:
Monaco Editor's _inlineComments_ are not highlighted / coloured as should (like when write to own line)
It seems to be klipper config problem.

As you can see, also values aren't highlighted when using : not =  (maybe not issue?)

https://i.imgur.com/kqJIePz.png

cadriel, 2021-07-04:
This screenshot is not Fluidd?

[image]

MrAntza99, 2021-07-04:
https://i.imgur.com/xtpnQMK.png
https://i.imgur.com/UeHjqQQ.png

Its Fluidd

cadriel, 2021-07-04:
The first screenshot you sent has the incorrect theme applied based on the latest release of Fluidd. I notice that your screenshots above are also for an old release of Fluidd.

It might be best if you were to update to v1.15 first - and see if your issue still persists.

The screenshot I posted above, which is directly from fluidd v1.15 - has commenting working correctly.

MrAntza99, 2021-07-04:
It was just too old version, it seems to work now well.

closed 2021-07-04 by MrAntza99