fluidd-core/fluidd #352: Erratic Current Layer Information until print_stats.print_duration > 0

truckershitch, 2021-07-05:
This is a very small thing.

The new layer info that you added in one of the last commits -- it's great to have!  I have been working on a printer-monitor Moonraker port and I used your quick calculation.  Moonraker seems to feed some erratic info until the print job actually begins (print_stats.print_duration == 0).  When it starts, your calculation works.

I don't know if this is a Moonraker issue or Klipper or what -- I have not looked -- it is July 4! 

Just an FYI for you.  And thank you so much for putting this together -- Fluidd runs beautifully.

cadriel, 2021-07-06:
Hi! Thanks for reporting. The next release of Fluidd has a new status widget, and the logic around this has changed a bit. I suspect it's no longer an issue - but please let me know if it still causes problems for you once 1.16 is out. :)

cadriel, 2021-07-08:
This is now release in 1.16.0! Please comment here if not resolved! thanks!

closed 2021-07-08 by cadriel