fluidd-core/fluidd #353: ADXL Support in Fluidd

MarcPot, 2021-07-07:
Since Klipper introduced input shaper we're all using adxl's to calibrate this function.

But the proces to do so is really tedious.

1. Calibrate
2. ssh in
3. use script to create images
4. SFTP in
5. Copy image to pc
6. open image
7. Repeat steps 1-6 until you're happy with the result
8. Save wanted values in klipper.cfg

Would be an awesome QOL improvement if we could do this through Fluidd just like the height map.

cadriel, 2021-07-08:
Hey, this is a decent idea - and is on the list, but is behind some other priorities for the moment. Thanks!

matmen, 2022-03-23:
Closing this because of staleness, feel free to re-open if this is still an ongoing issue.

MarcPot, 2022-03-23:
@matmen Sorry for not responding sooner.
Pedro his comment actually had not much influence on the topic, `SHAPER_CALIBRATE` already existed when I first opened this feature request but only comes after what I described

`SHAPER_CALIBRATE` doesn't show anything about the printer, so when actually tuning it you want to use the manual process, because then you can see the result of your tuning and if more tuning is needed. (Things like belt tension have a lot of influence on input shaper.)

`SHAPER_CALIBRATE` Just uses the settings it thinks is best but doesn't do anything about potential hardware issues.

matmen, 2022-03-24:
Considering this requires installing packages as well as running shell scripts, I don't think this is feasible at this point. Fluidd has no means of running any shell commands, so support in Moonraker is needed for this (as you already stated in your issue description). I'll move this to the backlog and I'll let you know if something changes (i.e. Moonraker adds this as a feature). Thanks for your FR!