fluidd-core/fluidd #355: Camera-Url ignored / override

Aeronaut, 2021-07-07:
fluidd: v1.15.0-7ae6ea5

## Affected Version(s):
- klipper version: all
- moonraker version: all
- fluidd version: all

## Platform:
Unrelated

## Hardware Configuration:
Unrelated

## Reproduction Steps:
1. Setup mjpegstreamer to stream 2 Cameras in one instance
2. Add Cameras in Fluidd with correct URI - e.g.: "/webcam/?action=stream_0" "/webcam/?action=stream_1"

## Actual Results:
On Dashboard the action-parameter of URI is replaced by "stream" / "snapshot" and it will show Camera 0 twice.

## Screenshot(s):
None

## Expected Results:
Both Cameras on Dashboard 

## Log Files:
Unrelated

## Additional information
I thing the error comes from [CameraItem.vue](https://github.com/cadriel/fluidd/blob/master/src/components/widgets/camera/CameraItem.vue#L149#L156) Line 149 and 156.
?action=stream_0 will get "action=stream" there if I understand right.

cadriel, 2021-07-08:
Ah, thanks for spotting. I'll sort this for the next release.

cadriel, 2021-07-08:
Ok, this is fixed in dev. Should be in the next release. Currently the fix I've applied is to not overwrite said query params - but in the future I'll extend the camera settings UI to better handle multi-cam setups too.

cadriel, 2021-07-08:
This is now release in 1.16.0! If not resolved, please comment here. :)

closed 2021-07-08 by cadriel

AnTeBiOss, 2021-09-29:
> This is now release in 1.16.0! If not resolved, please comment here. :)

I guess for me it is not solved with 1.16.0 :/