fluidd-core/fluidd #363: Gcode preview (beta) not showing whole printing area

Afterschnitzel, 2021-07-15:
Hi, 
I have V-Core3 300x300 and I have printing area 310x310mm.
I sliced object 300x200cm large in Prusaslicer.

Gcode preview is not showing whole printing (just 29x29cm) area and the left edge of the object in preview is far from the edge - in reality it is 0.5cm from the edge.

[image]

pedrolamas, 2022-03-03:
@Afterschnitzel any chance that you can provide us with the gcode file you used here so we can test on our side?

matmen, 2022-03-28:
Closing because of staleness. Feel free to re-open if this is still an ongoing issue.

corrado-c, 2022-05-01:
I also noticed this defect,
from me the decentralization is greater,
i think it depends a lot on how far endstop is from x0y0,.
it is done with any print.

`[stepper_x]
rotation_distance: 40  
microsteps: 32
step_pin: PF0
dir_pin: PF1
enable_pin: !PD7
endstop_pin: ^!PE5
position_endstop: -66
position_min: -66
position_max: 230
homing_speed: 40

[stepper_y]
rotation_distance: 40
microsteps: 32
step_pin: PF6
dir_pin: PF7
enable_pin: !PF2
endstop_pin: ^!PJ1
position_endstop: -43
position_min: -43`

pedrolamas, 2022-05-01:
Thanks @corrado-c, can you confirm if you are experiencing this on the latest release 1.18.0?

corrado-c, 2022-05-01:
> Thanks @corrado-c, can you confirm if you are experiencing this on the latest release 1.18.0?
yes i tried right after i updated everything,
I noticed that sometimes the zoom works, almost always does not, but the image always remains cropped

[image]
[image]
[image]

corrado-c, 2022-05-11:
[image]

now it works .... did any other update affect this problem?

corrado-c, 2022-05-11:
I understood where the problem is,
if in the settings, I activate FLIP IMAGE, vertical or horizontal, the image is then badly cropped.
if I deactivate FLIP IMAGE everything is ok.

pedrolamas, 2022-05-26:
@corrado-c I think you are on the right track here, it seems that having negative `position_min` values is causing this, at least partially!

My knowledge of the gcode previewer code is quite limited, but I have started a branch with a possible fix: https://github.com/pedrolamas/fluidd/tree/pedrolamas/363

I still need to test this further, but it would help if you can provide your full printer.cfg and a sample file where you see this issue hapening!

matmen, 2022-07-07:
@corrado-c Hey, any chance you could provide your printer config and a sample gcode file? Would love to get this fixed in 1.19 😉

closed 2022-08-08 by pedrolamas