fluidd-core/fluidd #365: Gcode Preview is out of sync with the printer by several minutes

NameOfTheDragon, 2021-07-17:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: v0.9.1-617-gf2421005
- moonraker version: v0.7.1-25-g7f740cc
- fluidd version: v1.16.2-0fa3c9b (from app.fluidd.xyz)

## Platform:
Klipper host: Raspberry Pi 3B+
Browser: Edge on Windows 10 [Version 91.0.864.67 (Official build) (64-bit)]

## Hardware Configuration:
Voron 2.4 350x350
Raspberry Pi 3B+ (no camera)
2 x BTT SKR 1.4 Turbo

## Reproduction Steps:
1. Print something
2. Load the file into gcode preview
3. Check "Follow Progress".

The following file was printed.
[Faker_Knife.zip](https://github.com/cadriel/fluidd/files/6834693/Faker_Knife.zip)

## Actual Results:
Gcode Preview "Follow Progress" mode is out of sync with the printer by several (perhaps as much as 10) minutes.
A little bit of lag is to be expected, but not 10 minutes. I don't think this can be explained away by Klipper's look-ahead buffer because pause/resume works within a few seconds.

## Screenshot(s):
Timing issue, screen shot probably not useful, but here it is anyway:
[image]

## Expected Results:
Display should track printer activity within a reasonable time window. A few seconds of lead or lag is to be expected,
10 seconds of lead or lag is problematic, 10 minutes of lag is just broken.

## Log Files:
(The log files fluidd provides at Configure > System Control.)
- klipper log:
- moonraker log:
** I don't see log files under Configure, only "Configuration files" and "Other files" which has sub-categories "Docs" and "Config examples" **

StuSerious, 2021-07-19:
hey there @NameOfTheDragon!

thank you for your report on the Preview!
this was already known, we are tracking these issues to the source, we'll update here when it will get fixed

cheers! 
_~S_

NameOfTheDragon, 2021-07-20:
As an extra data point, I believe the display is ahead of the printer and not lagging behind it.

Rian-Striegel, 2021-07-20:
Just for info: I agree on the gcode preview being out of sync (always ahead more and more during the job). For the display, it has always been a little ahead of the actual gcode, was in Marlin for me and is the same for klipper. I always assumed it to be buffering in the MCU and it never bothered me. This leading is small and constant during the job.

NameOfTheDragon, 2021-08-12:
More information just in. The gcode preview doesn't even agree with the print progress display about the number of layers, let alone which one it's on... According to SuperSlicer, 787 layers is the correct number, and the gcode preview has this correct. But note the print progress display has the wrong layer number and the wrong total number of layers. This screen shot was taken on layer 787 of 787.

[image]

Hajto, 2021-08-14:
I am facing the same issue:
fluidd v1.16.2
klipper v0.9.1-676-gf759df3c
moonraker v0.7.1-35-g5204fe3

pedrolamas, 2022-02-20:
Fixed by #469

closed 2022-02-20 by pedrolamas