fluidd-core/fluidd #369: Macros panel seemingly randomly opens different categories

NameOfTheDragon, 2021-07-23:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: v0.9.1-617-gf2421005
- moonraker version: v0.7.1-25-g7f740cc
- fluidd version: v1.16.2-0fa3c9b (from app.fluidd.xyz)

## Platform:
(Information about your platform, like the OS or browser you are using.)
Windows 10, Microsoft Edge browser

## Hardware Configuration:
(Information about your hardware configuration. Like MCU, host and setup)

## Reproduction Steps:

Scenario 1:
1. Open Fluidd by browsing to http://app.fluidd.xyz - a random macro category will be expanded

Scenario 2:
1. Start with the Fluidd home screen open.
2. Switch to another screen (e.g. Settings)
3. Switch back to the Home screen - a random Macro category will be expanded.

## Actual Results:

Fluidd seemingly randomly expands one or more macro categories. Not every time, but there is a nonzero chance of this happening. There is no consistency about which categories will be displayed. It just seems to pick one (sometimes several) at random.

If I arrange the display of macros just as I want it, then switch to another page, e.g. the Settings page, then when I switch back to the home page, SOMETIMES it will have kept my preferences intact, but there is a nozero chance that some different random set of macros will be visible.

Just opening the app by browsing to the URL sometimes shuffles the macros.

## Screenshot(s):
I made [a video of the issue to prove I'm not going crazy](https://youtu.be/uik8Ow0_BvI) ;-)
https://youtu.be/uik8Ow0_BvI
(video is public but unlisted on YouTube)

## Expected Results:
I would want Fluidd to remember which macro categories I had expanded and keep that set until I change it.
If that's not possible, then I'd like to be able to specify (in settings) the categories that I want shown by default.

matmen, 2022-03-16:
Hey, I just tried to reproduce it but couldn't get the same behavior. Does this still happen on 1.17.2 with latest Moonraker?

matmen, 2022-03-23:
Closing this because of staleness, feel free to re-open if this is still an ongoing issue.

closed 2022-03-23 by matmen