fluidd-core/fluidd #373: [v1.16.2] Printer statistics do not reset

qSib, 2021-07-27:
When I press "remove all" under history and confirm, my stats will all be zeroed out. But as soon as I reload / force refresh / clear site cookies, all old stats are still there

## Affected Version(s):
- klipper version: v0.9.1-632-g7e88f922
- moonraker version: v0.7.1-33-g86e0ef0
- fluidd version: v1.16.2

## Platform:
Firefox, Chrome

## Hardware Configuration:
RPi Zero (non-W) (using FluiddPI image) connected over serial

## Reproduction Steps:
1. Press "remove all" under history and confirm.
2. Reload / force refresh / clear site cookies

## Actual Results:
Old statistics will still be there, while all past prints are gone.

## Screenshot(s):
[image]

## Expected Results:
Statics should be zeroed out.

## Log Files:
- klipper log:

```
Starting Klippy...
Args: ['/home/pi/klipper/klippy/klippy.py', '/home/pi/klipper_config/printer.cfg', '-l', '/home/pi/klipper_logs/klippy.log', '-a', '/tmp/klippy_uds']
Git version: 'v0.9.1-632-g7e88f922'
CPU: 1 core ARMv6-compatible processor rev 7 (v6l)
Python: '2.7.16 (default, Oct 10 2019, 22:02:15) \n[GCC 8.3.0]'
Start printer at Tue Jul 27 16:24:37 2021 (1627395877.8 57.1)
===== Config file =====
[virtual_sdcard]
path = ~/gcode_files

[display_status]

[pause_resume]

[stepper_x]
step_pin = PC2
dir_pin = PB9
enable_pin = !PC3
microsteps = 16
rotation_distance = 40
endstop_pin = ^PA5
position_endstop = 0
position_min = 0
position_max = 246
homing_speed = 50

[stepper_y]
step_pin = PB8
dir_pin = PB7
enable_pin = !PC3
microsteps = 16
rotation_distance = 40
endstop_pin = ^PA6
position_endstop = 0
position_min = 0
position_max = 231
homing_speed = 50

[stepper_z]
step_pin = PB6
[... 495 lines cut ...]
Branch: master
Remote URL: https://github.com/Arksine/moonraker.git
Current Commit SHA: 86e0ef0d5eb80742e74943d752b002c89bad1596
Upstream Commit SHA: 86e0ef0d5eb80742e74943d752b002c89bad1596
Current Version: v0.7.1-33
Upstream Version: v0.7.1-33
Is Dirty: False
Is Detached: False
Commits Behind: 0
2021-07-27 16:25:23,120 [base_deploy.py:log_info()] - Application moonraker: Channel: dev, Need Channel Update: False
2021-07-27 16:25:23,121 [base_deploy.py:log_info()] - Application moonraker: Validity check for git repo passed
2021-07-27 16:25:23,366 [update_manager.py:github_api_request()] - GitHub API Request Processed
URL: https://api.github.com/repos/cadriel/fluidd/releases
Response Code: 200
Response Reason: OK
ETag: "27f3fe7573f0c86652c057e8576cdbb4674c56d97f87a3551d03047d4cbf4657"
2021-07-27 16:25:23,402 [update_manager.py:_get_remote_version()] - Github client Info Received:
Repo: fluidd
Local Version: v1.16.2
Remote Version: v1.16.2
Pre-release: False
url: https://github.com/cadriel/fluidd/releases/download/v1.16.2/fluidd.zip
size: 9518655
Content Type: application/zip
```

Biorn1950, 2022-01-28:
Hello,, this is/was normal. This function just clear the history list and not the global stats.
A new endpoint in moonraker has been added to delete the global stats. So from now it's possible to add a button for that.

qSib, 2022-01-28:
Thanks for the clarification. That makes sense.

closed 2022-01-28 by qSib

Biorn1950, 2022-01-28:
Ha but the message is not good. I understand now. I'm going to add 2 different buttons.

Biorn1950, 2022-01-28:
@qSib voila
[image]