fluidd-core/fluidd #377: Can I install fluidd at other hardware，such as Allwin H6 with Armbian？

dgpower, 2021-08-03:
Raspberry Pi is too expensive. I want to use a TV box to install fluidd, it uses Allwin H6, armbian system.

How can I manually install and run fluidd?

nlef, 2021-08-05:
Maybe [kiauh](https://github.com/th33xitus/kiauh) may help

dgpower, 2021-09-05:
谢谢！我现在确实是用这个脚本安装的。很方便！

可惜这个脚本其中一段还继续用Pythion2.7,。最好能改为Python3就完美了

抄送人：Special ***@***.***>,  Author ***@***.***>
主题：Re:_[cadriel/fluidd]_Can_I_install_fluidd_at_other_hardware，such_as_Allwin_H6_with_Armbian？_(#377)
日期：2021年08月06日 05点20分

&mdash;

Ainmimdal, 2021-09-06:
I installed fluidd on a rockhip tv box using kiauh. Installs fine

pedrolamas, 2022-02-18:
Yes, you should be able to install in most recent Debian derivatives with little to no issues at all - and that includes Armbian!

Alternatively, if your system can run Docker, you can then just use the Docker image and run that!

closed 2022-02-18 by pedrolamas