fluidd-core/fluidd #380: Select/Copy/Paste doesn't work on iOS when editing config files

russiancatfood, 2021-08-04:
Latest Fluidd build won't let me select/copy/paste bits of config when using the build-in editor.

This worked in earlier builds, before the more fancy VCCode style editor was added.

StuSerious, 2021-08-06:
hey there @russiancatfood!

thanks for pointing that out! 
sadly there is not much that can be done about this bug, since it's not caused by fluidd. 
we are going to have to wait for upstream fixes before we can do anything..

I'll ping you here and let you know when that changes, it will also defenetely be in the fluidd release changelog when they manage to fix it :P

Cheers!
_~S_

russiancatfood, 2021-08-08:
Thank you!
I figured, but I didn't know where to file it :)

Looking forward to getting this to work again. I seem to be one of the few mad men doing initial printer bring-up using a phone.

matmen, 2022-02-25:
Hey, according to the [webkit bug tracker](https://bugs.webkit.org/show_bug.cgi?id=210071) this should have been solved with iOS 14.5, can you confirm?

susisstrolch, 2022-02-26:
@matmen it neither works on iPad OS 15.3 nor iPhone OS 15.3. 
Also the cursor move by space key doesn’t work.

PR #649 (2022-04-29): fix: show simple editor on mobile
Monaco Editor is not supported on mobile devices (it says so on [their page](https://microsoft.github.io/monaco-editor/)), so instead of having a broken editing experience in mobile devices, better to just use a simple text area!

This change can be easily tested by changing the browser User Agent to a mobile one.

Screenshot from my own OnePlus Android phone:

[image]

Fixes #380
closed 2022-04-29 by pedrolamas