fluidd-core/fluidd #381: Cant get rid of "CANCEL_PRINT macro not found in configuration warning"

sgall17a, 2021-08-06:
## Affected Version(s):
fluidd.  v1.16.2
UP TO DATE
klipper v0.9.1-667-g31761500
UP TO DATE
moonraker. v0.7.1-34-g5522e15

Raspberry Pi Zero W Rev 1.1
--
ARMv6-compatible processor rev 7 (v6l)
358.0 MB.  BCM2835
32bit armv6l with 1 cores
Raspbian GNU/Linux 10 (buster)
buster. debian. v0.9.1-667-g31761500

(The versions of fluidd this issue occurs on.)
- klipper version:
- moonraker version:
- fluidd version:

## Platform:
Macbook ->Chrome -> 

## Hardware Configuration:
Pi Zero-> Mks mini e3-> ender3 pro

## Reproduction Steps:
Installed fluidd.zip package on raspberry pizero and updated from web interface.

2. Connected pi zero and ender3 with klipper firmware from package above.
3. Everything seems to work well (x,y,z movements, bed and extruder heater ok)
4. Add rows in fluidd.config as per https://docs.fluidd.xyz/configuration/initial_setup

## Actual Results:
Keep getting 
"CANCEL_PRINT macro not found in configuration. even though it is there as per documents."
The other warnings from before adding the extra rows have gone away.

## Expected Results:
When I define CANCEL_PRINT macro the warning should go away.

Will my prints mysteriously fail because of this warning?:

m-sundman, 2021-08-09:
Based on some comments I found by googling the problem seems to be related to this in `fluidd.cfg`:
```
[gcode_macro CANCEL_PRINT]
...
rename_existing: CANCEL_PRINT_BASE
...
```
Can't add another CANCEL_PRINT macro, presumable because it exists in `fluiff.cfg`, but it still shows that warning about CANCEL_PRINT missing. I wouldn't be surprised if this was the cause of the position confusion in issue #385 tho.

RolledR, 2021-08-16:
I'm having the same issue on FluiddPi, except mine complains about `[virtual_sdcard]`,`[pause_resume]`,`[display_status]`, AND `CANCEL_PRINT`, despite copying everything from Initial Setup to `fluidd.cfg`.

I tried deleting `fluidd.cfg` as well as restarting both the printer and the pi.
Pi and Fluidd fully updated.
fluidd: v1.16.2
klipper: v0.9.1-679-g7fddc5a8
moonraker: v0.7.1-35-g5204fe3

My [fluidd.cfg](https://github.com/cadriel/fluidd/files/6989355/fluidd.cfg.txt) (.txt so I could upload it.)

sgall17a, 2021-08-16:
Hi,
I fixed my  problem by putting the cancel_print and other macros in printer.cfg.
I think it was my silly mistake and possibly slightly unclear documentation.

RolledR, 2021-08-17:
Thanks, that fixed it for me as well. Not quite sure what's going on then, as I was following [this](https://old.reddit.com/r/VoxelabAquila/comments/ovk7t2/howto_klipperize_your_aquila/) tutorial, and it seemed like everyone else did it as I did without running into any issues.

I would appreciate some clarification about whether the tutorial is wrong or this is a bug. Or if I just can't follow instructions and missed something.

closed 2022-01-29 by Biorn1950

justheath, 2022-02-06:
Does your `printer.cfg` import your `fluidd.cfg`?
If not, add this:
 `[include fluidd.cfg]`?

insanie, 2025-11-24:
Was fighting this thing too. It ended up with this piece of documentation: https://www.klipper3d.org/G-Codes.html?h=pause_resume#pause_resume
Basically, you first need to put plain `[pause_resume]` into your config so that all pause/resume - and cancel! - controls are enabled in klipper.