fluidd-core/fluidd #383: Hot folder for uploading prints

buikpijn, 2021-08-08:
## Is your feature request related to a problem? Please describe.
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

## Describe the solution you'd like:
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

## Describe alternatives you've considered:
(A clear and concise description of any alternative solutions or features you've considered. Again, if possible, think about why these alternatives are not working out.)

## Affected users and/or setups:
(Who do you think will benefit from this? Is everyone going to benefit from these changes? Or specific kinds of users?)

## Additional Context:
(Add any other context or screenshots about the feature request here.)

buikpijn, 2021-08-08:
If it existst i am worse in searching than i though.. I'm unable to find any information in regards to my request;

I'd like a watched folder where you can upload prints to.

I tried this through the standard virtual sd-card folder, creating a SMB share to put files in that folder, but Fluidd can't read the meta-data of files that are placed in there and it also threw an SD busy error.

I know there is a octoprint spoofer for uploading from slic3r, but s3d is my goto slicer.

buikpijn, 2021-08-16:
How do you upload your prints? Always through the web-interface?

Thanks in advance for your response.

matmen, 2022-02-22:
Requires support in [fluidd-core/FluiddPI ](https://github.com/fluidd-core/FluiddPI) for SMB shares first

pedrolamas, 2024-03-04:
We have stopped supporting FluiddPi a long time ago, so this has gone outside of our scope - as Fluidd is just a web interface that runs on the client browser.

Having said that, Moonraker already monitors the `gcodes` folder (by default, `~/printer_data/gcodes``) and Fluidd will automatically show any new file that gets created there.

pedrolamas, 2024-03-04:
One final note here I completely almost forgot: Fluidd support drag & drop of files AND folders directly to the browser (files will just be uploaded), so that might also be a nice way to go around this!

closed 2024-03-04 by pedrolamas