fluidd-core/fluidd #385: Not even relative moves are allowed without homing

m-sundman, 2021-08-09:
## Affected Version(s):
- klipper version: v0.9.1-670-gfa7a91a3
- moonraker version: v0.7.1-34-g5522e15
- fluidd version: v1.16.2

## Platform:
Raspberry Pi, linux

## Hardware Configuration:
Raspberry Pi 3B

## Reproduction Steps:
1. Cancel a print.

## Actual Results:
Fluidd refuses to let me move the nozzle, which is stuck to the print.

## Screenshot(s):
[image]

## Expected Results:
If fluidd gets confused about the exact position then it should still allow relative motion, so I can lift the nozzle, so I can remove the print, so I can home the axis. If not elseway then at least using the console. Or by some kind of override option.

## Log Files:
- klipper log: (empty file)
- moonraker log: [moonraker.log](https://github.com/cadriel/fluidd/files/6957758/moonraker.log)

## Additional information
It's OK if fluidd would warn before moving the head when it's not sure where the head is. But it must allow it. Requiring me to power off motors and turning them by hand is quite inconvenient with certain setups.

ChipCE, 2021-08-29:
This is how klipper work, not fluidd bug. You can move axis by enable force_move
https://www.klipper3d.org/Config_Reference.html#force_move

m-sundman, 2021-08-30:
> This is how klipper work, not fluidd bug. You can move axis by enable force_move
> https://www.klipper3d.org/Config_Reference.html#force_move

The nozzle getting stuck in the print is clearly not what anyone would want by default, so would be considered a bug. I don't know if it's in fluidd or klipper, but fluidd is the UI, and there seems to be no clear way of doing this in the UI, so I'd go with it being at least partly fluidd.
Or how would I go about making it work? Let's say I `enable_force_move: True` in the klipper conf...and then what?

ChipCE, 2021-08-30:
> The nozzle getting stuck in the print is clearly not what anyone would want by default,

I think the same as not allow to move without homing, not everyone want the toolhead/bed to hit the frame/other parts due the printer doesn't known the current position of the axis.

> Or how would I go about making it work? Let's say I `enable_force_move: True` in the klipper conf...and then what?

https://www.klipper3d.org/G-Codes.html#force-movement

m-sundman, 2021-08-31:
> > The nozzle getting stuck in the print is clearly not what anyone would want by default,
> 
> I think the same as not allow to move without homing, not everyone want the toolhead/bed to hit the frame/other parts due the printer doesn't known the current position of the axis.

When I cancel a print I'm not telling fluidd to forget where the toolhead/bed is, nor am I telling it to leave the nozzle stuck in the middle of the print. Couldn't fluidd do something smarter than that?

> > Or how would I go about making it work? Let's say I `enable_force_move: True` in the klipper conf...and then what?
> 
> https://www.klipper3d.org/G-Codes.html#force-movement

So where should I put what so I can control this from fluidd?

ChipCE, 2021-08-31:
> When I cancel a print I'm not telling fluidd to forget where the toolhead/bed is, nor am I telling it to leave the nozzle stuck in the middle of the print. Couldn't fluidd do something smarter than [...]

There is CANCEL_PRINT macro, you can change that so the fw not disable the motor after print so it will remember the position of the axis. You can also use move command if motor is not disabled after each print.

> So where should I put what so I can control this from fluidd?

Not the best way but you can set it via gcode_macro, it will add a button into fluid web UI. https://www.klipper3d.org/Config_Reference.html#gcode_macro

Go to discord, You will get faster respond and example from ppl there.

danijoo, 2021-10-04:
Wouldn't it be possible to show a switch in the Fluidd Tool section that - when turned on - uses force_move when a command is send to klipper and uses a normal move otherwise?

Biorn1950, 2022-01-29:
> Wouldn't it be possible to show a switch in the Fluidd Tool section that - when turned on - uses force_move when a command is send to klipper and uses a normal move otherwise?

@danijoo I like this idea. I'm going to make a feature request.

@m-sundman hopefully it's how klipper work, if the printer don't know the position no movement will make. You could break all if you move out of range. For your issue you can edit the cancel_print like already mentioned.

closed 2022-01-29 by Biorn1950