fluidd-core/fluidd #396: Circular Bed Mesh Visualisation

IanFluidd, 2021-08-29:
After I do a BED_MESH_CALIBRATE on my **delta** printer, the mesh from the bed probe is displayed in the 'Tune' tab as it always is. However, I've found that when displaying the mesh data from a round bed, Fluidd still shows a square plane, not only making it incredibly hard to understand where exactly the visualised data is referring to on the actual bed, but also adding to the displayed probed points, where no point was probed.

For example, the default round bed for a printer running klipper is probed in the following locations (indicated by x):
.....x
..x x x
x x x x x
..x x x
.... x

However, the generated mesh shows the probed points in these locations (a square):
x x x x x
x x x x x
x x x x x
x x x x x
x x x x x

So where do these extra points, represented by 'a', come from? Are they interpolated? This doesn't make sense since there is physically no bed there.
a a x a a
a x x x a
x x x x x
a x x x a
a a x a a

I would like an alternate mesh visualisation that applies automatically to beds defined as circular (and probed as such) that displays an accurate, CIRCULAR mesh in the 'Tune' menu instead of the current square one. This will make meshes of circular beds much easier to understand and provide clearer feedback.

I'm not sure why circular bed meshes have always been displayed as a square plane, and perhaps there is a reason for this that I haven't yet thought of, but it really seems counterintuitive for something that is meant to help users understand exactly how their beds look like to appear as a completely different shape to what it actually represents.

This new circular bed mesh visualisation will of course benefit all 3d printer users running Klipper and Fluidd who's printers have circular beds (so mainly the traditional delta community, but also rotary delta users among others). 

It would be much appreciated if you took the time to implement this feature - The Delta Community

matmen, 2022-02-22:
Related: #401

manu69x, 2024-05-30:
Please revive this request

IanFluidd, 2024-05-30:
Has this still not been addresed? I've been away from 3D printing for quite a while now (busy with Uni), but I expected this to be a bigger priority.  I guess delta users are a smaller minority then I previously thought.

To any devs reading this we still love you guys!

manu69x, 2024-05-31:
Many people like me now have the opportunity to upgrade their old 8-bit delta printers with this fantastic klipper program that we are all excited about, at a cost of around $20 for an amlogic s905w2 or s905y4 tvbox with all the power of a 32 bit (equal to raspi 3b+). It's a real shame that fluidd doesn't display the circular plate mesh correctly. Small differences in the length of an arm have repercussions on a non-linear z movement of the head and this can be seen very well if there was the possibility of correctly visualizing the circular mesh.

Thanks for your hard work

pedrolamas, 2024-06-05:
I've taken a quick look into this, and believe there is some confusion here: while the bed on delta printer is indeed circular, the mesh from it is still a square, just as it is shown in the [Klipper documentation](https://www.klipper3d.org/Bed_Mesh.html#round-beds) (highlighted in orange below):

[image]

There is a big possibility the orientation is incorrectly shown (45 degrees rotated in Z), but nevertheless, the representation should be accurate.

Now I personally don't have Delta printer to test and confirm any of this, so we will need someone from the community to contribute any improvements on this area.