fluidd-core/fluidd #398: [1.16.2] Power Button from Moonraker out of sync

multiholle, 2021-09-01:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: 0.9.1-763-g3a497d04
- moonraker version: 0.7.1-42-g81176a6
- fluidd version: 1.16.2

## Reproduction Steps:
- setup a smart outlet power switch for the printer in moonraker
- configure klipper to switch off power when idle
- wait for idle timeout and printer shutdown

printer.cfg
```
[idle_timeout]
timeout: 300
gcode:
  M84
  TURN_OFF_HEATERS
  UPDATE_DELAYED_GCODE ID=delayed_printer_off DURATION=60

[delayed_gcode delayed_printer_off]
initial_duration: 0.
gcode:
  {% if printer.idle_timeout.state == "Idle" %}
    POWER_OFF_PRINTER
  {% endif %}
```
moonraker.conf
```
[power homeassistant_switch]
type: homeassistant
address: 192.168.178.90
port: 8123
device: switch.bed_desk_docking_station
token: topsecrettoken
domain: switch
restart_klipper_when_powered: True
restart_delay: 5
```

## Actual Results:
The power menu in fluidd (top right corner) will show the printer as "on" but the printer is actually turned off.

## Expected Results:
The power menu in fluidd (top right corner) will show the printer as "off".

vajonam, 2021-11-13:
yeah not sure whats going on here.. but double clicking seems to get things back in sync.. is the intended behaviour?

ForsakenHarmony, 2021-11-15:
I'm assuming reloading the page doesn't fix it?

multiholle, 2021-11-15:
Reloading the page doesn't help

vajonam, 2021-11-15:
Here is gif, the screen on the right is homeassistant that accurately reflects the state of the button. 

The first click doesn't see to send any requests. I can confrim that moonraker side is good because when I call the GCODE that does the power off/on. it works the very first time. Idle timeout for example works fine. 

[image]

vajonam, 2021-11-15:
Both Mainsail and Fluidd have this issue, which made me thing it was moonraker thing, the code in there looks like it has all the bits implemented to query the state of the switch using the `/states/` API in homeassistant.

ForsakenHarmony, 2021-11-15:
Can you try going to `{moonraker_url}/machine/device_power/devices` and see which status is shown there?

vajonam, 2021-11-15:
Hmm. Your right.

```
{"result": {"devices": [{"device": "Ender3", "status": "off", "locked_while_printing": false, "type": "homeassistant"}]}}
```

This while the printer is connected and is on. Will open bug in moonraker.

ForsakenHarmony, 2021-11-15:
I'll close this, as the issue should be fixed when moonraker fixes it on their side

closed 2021-11-15 by ForsakenHarmony

vajonam, 2021-11-15:
I have found a provided a fix the code but issue was the return status needed a sleep for to give time for home assistant to return the correct state.

vajonam, 2021-11-15:
This fixed as per.
https://github.com/Arksine/moonraker/issues/273