fluidd-core/fluidd #402: [1.16.2] Bug? No durations logged in Job history.

CraXyOW3, 2021-09-02:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: v0.9.1-743-g95047784
- moonraker version: v0.7.1-42-g81176a6
- fluidd version: v1.16.2

## Platform:
Win10/ Firefox

## Hardware Configuration:
RPi 3b+ as host.

## Reproduction Steps:
Just printed

## Actual Results:
In history i get everything except the info that is in all the "Duration" columns.

## Screenshot(s):
[image]

## Expected Results:
Same list but with "Duration" columns filled with information.

## Log Files:
(The log files fluidd provides at Configure > System Control.)
- klipper log: [klippy.log](https://github.com/cadriel/fluidd/files/7100738/klippy.log)
- moonraker log: [moonraker.log](https://qwuille.eu/moonraker.log)

## Additional information
I was not able to upload moonraker.log as i did with klippy.log, the link is of my own place.

CraXyOW3, 2021-09-03:
Ok, more searching gave me an answer actually.
Looked into moonrakers config and saw the [history] block which in turn activated the history tab.
I recon that the duration columns on the "job" tab is times generated by slicers?

Biorn1950, 2022-01-29:
@CraXyOW3 can you close the issue. Thanks.

closed 2022-01-29 by Biorn1950