fluidd-core/fluidd #410: Feature request: User specific layouts

OneMoreByte, 2021-09-11:
## Is your feature request related to a problem? Please describe.
I use fluidd on both on a phone right next to my printer and my desktop. I don't want the camera and some other things visible on the phone. I created users expecting the layout to be specific to a user but updating the layout as `User1` also updated it for `User2`

## Describe the solution you'd like:
`User1` and `User2` have independent layouts when they log in. 

## Describe alternatives you've considered:
Layouts that are per device, maybe the settings could be saved on each machine.

## Affected users and/or setups:
I'm not sure who would benefit from this. I imagine its a weird edge case thing for _some_ people with multiple devices or for multi-user setups.

bulbersator, 2021-12-12:
I am in the same situation. I think the big advantage of Fluidd is that you can just use an old phone as a screen next to the printer, instead of buying and configuring a new screen for RPI. I think the most common way of working with Fluidd involves a desktop computer for the main setup and a phone / tablet as the screen next to the printer. Therefore, it would be very useful to be able to set the page layout for a user or a device.

matmen, 2022-02-22:
Related: #370

PR #878 (2022-09-29): feat: user/device specific layouts
Creates user and device specific layouts on layout change (based on concatenation of `dashboard-{user}-{device-size}`, clones current `dashboard` if no custom one exists yet).

Allows the overriding of the default layout for new users (or other devices), which can also be reset to the initial state.
Not 100% sure about the placement and usage of header buttons here, but it's what was there before:
[image]

Closes #370
Closes #410
closed 2022-09-29 by matmen