fluidd-core/fluidd #413: FR: Config Backups (For all *.cfgs -- not just printer.cfg)

bad1dea, 2021-09-14:
## Is your feature request related to a problem? Please describe.
I think the printer.cfg backup (with timestamp) feature should be implemented on all config files -- as most user cases now have split cfgs for more advanced klipper usage.

## Describe the solution you'd like:
Implement same printer.cfg backup/revision on all *.cfg files.

## Affected users and/or setups:
Users who have setup their klipper config files into individual files for easier editing/readability.

i.e 
printer.cfg -> stores all printer related variables/settings and loads other cfg files
cfgs/steppers.cfg -> stepper config
cfgs/extruder.cfg -> extruder config
etc

Biorn1950, 2022-01-29:
> I think the printer.cfg backup (with timestamp) feature should be implemented on all config files -- as most user cases now have split cfgs for more advanced klipper usage.

Hello, from my experience only ratrig users have this kind of structure and it's a nightmare when they ask for help.

Anyway it's a good idea. I'll going to make the backup optional and add all the .cfg files.

matmen, 2022-02-25:
Related: #311

matmen, 2022-03-03:
Unsure about whether this is in scope for Fluidd or not. The printer.cfg backups are all created by Klipper. @StuSerious @pedrolamas opinions?

closed 2022-03-03 by pedrolamas