fluidd-core/fluidd #414: Tapping on console commands does not work on Edge mobile

btastic, 2021-09-15:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: v0.9.1-763-g3a497d04
- moonraker version: v0.7.1-42-g81176a6
- fluidd version: v1.16.2

## Platform:
OS: Galaxy S10 with Android 11
Browser: Edge 46.06.4.5161

## Hardware Configuration:
- I guess not relevant

## Reproduction Steps:
1. Open fluidd in Edge
2. Tap on a command in the console

## Actual Results:
- Command does not get pasted into the input field

## Screenshot(s):
- No idea how to exactly show it

## Expected Results:
- Command gets pasted into the input field

## Log Files:
(The log files fluidd provides at Configure > System Control.)
- klipper log: <not relevant>
- moonraker log: <not relevant>

## Additional information
This seems to be an Edge mobile specific issue. I tried this on Firefox and it works there.
Edge desktop seems fine too.

pedrolamas, 2022-03-10:
I just tested this in my own phone (OnePlus 9 Pro running latest Edge dev) and it worked fine!

@btastic can you please check with the latest version?

pedrolamas, 2022-03-14:
Closing as it seems to be working correctly (might have been an issue in an older version of Edge or we just updated a dependency that fixed the problem).

Please feel free to re-open if you are still experiencing this issue.

closed 2022-03-14 by pedrolamas