fluidd-core/fluidd #419: Unable to send Gcode from SuperSlicer to Fluidd v1.16.2

Exit2Studios, 2021-09-27:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: v0.9.1-792-g31ac9dc6
- moonraker version: v0.7.1-59-g81b0c0a
- fluidd version: v1.16.2

## Platform:
RaspberryPi 3b+, Raspbian GNU/Linux 10 (buster)

## Hardware Configuration:
RaspberryPi 3b+, Raspbian GNU/Linux 10 (buster)

## Reproduction Steps:
1. Setup super slicer per instructions to connect to physical printer.  Test button successful
2. Get error

## Actual Results:
Error uploading to print host:
HTTP 400: {"error": {"code": 400, "message": "Bad Request", "traceback": "Traceback (most recent call last):\n\n File \"/home/pi/moonraker/moonraker/app.py\", line 804, in post\n result = await self.file_manager.finalize_upload(form_args)\n\n File \"/home/pi/moonraker/moonraker/components/file_manager/file_manager.py\", line 419, in finalize_upload\n upload_info = self._parse_upload_args(form_args)\n\n File \"/home/pi/moonraker/moonraker/components/file_manager/file_manager.py\", line 444, in _parse_upload_args\n raise self.server.error(f\"Root {root} not available\")\n\nutils.ServerError: Root gcodes not available\n\n\nDuring handling of the above exception, another exception occurred:\n\n\nTraceback (most recent call last):\n\n File \"/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py\", line 1704, in _execute\n result = await result\n\n File \"/home/pi/moonraker/moonraker/app.py\", line 807, in post\n e.status_code, str(e))\n\ntornado.web.HTTPError: HTTP 400: Bad Request (Root gcodes not available)\n"}}

## Screenshot(s):
(Images or gifs showing the problem, perhaps before/after.) 

## Expected Results:
(What *should* happen after the reproduction steps have been followed.)
Gcode should upload to Fluidd
## Log Files:
(The log files fluidd provides at Configure > System Control.)
- klipper log: empty?
- moonraker log:
[moonraker (1).log](https://github.com/cadriel/fluidd/files/7238075/moonraker.1.log)

## Additional information

Exit2Studios, 2021-09-27:
[image]

komandrik, 2021-09-28:
> Error uploading to print host:
In the PrusaSlicer printer setting, in the API KEY field, enter the key from the moonraker access setting

according to the screenshot - check the path to the folder with the files

steinerlein, 2021-12-19:
I'm having the same issue. Were you able to find a remedy?

Biorn1950, 2022-01-29:
For moonraker related issue you can use the [Klipper discord](https://discord.gg/ctSg5BpB) or the [Klipper discourse](https://klipper.discourse.group/)

closed 2022-01-29 by Biorn1950