fluidd-core/fluidd #420: implement LDAP authentication

yagarea, 2021-09-28:
## Is your feature request related to a problem?
I have central authentication system using LDAP. I would like to be able to use it to authenticate to fluidd interface

## Describe the solution you'd like:
Having LDAP implemented so I can redirect it to my authentication server and use it for login.

shinuza, 2022-01-30:
Hi there. This would require work in the upstream project namely [moonraker](https://github.com/Arksine/moonraker). If you get them to implement LDAP then we could implement the UI for it, in the meantime I'm closing this.

PS: You could also consider an API gateway like [Kong](https://konghq.com/kong/)

smhaller, 2022-02-28:
I put this here because I stumbled upon this entry - maybe this helps someone...
I think this should not too much work in moonraker (but if you want to do this for the moonraker project in general  this needs to be way more elaborate):
I think for a quick fix all you have to do is to adapt moonraker/components/authorization.py
  - Install python ldap
 - **Note**: this is an untested stub (it first tries to authenticate against ldap and then against the moonraker db).

     
```python
import ldap

def check_ldap_credentials(user_name, passwd):
   """Verifies ldap credentials for username and password.
   return: true on success, false on failure
   """
   server="yourldapserver.com"
   user = "uid="+user_name+",ou=people,dc=domain,dc=com"
   try:
     connect = ldap.open(server)
     connect.bind_s(user,passwd)
     return True
   except ldap.INVALID_CREDENTIALS:
     connect.unbind()
     return False
   except ldap.LDAPError:
     connect.unbind_s()
     return False
...
# insert after line 368 a check 
ldapauthenticated=False
if check_ldap_credentials(username,password):
  ldapauthenticated=True
  # create user in db if it does not exist - otherwise proceed to db auth
  if ! username in self.users:
    create=True
else:
  raise self.server.error("Invalid Password") 

...
#Line 392
  if (hashed_pass != user_info['password']) and not ldapauthenticated:

lucaschoeneberg, 2022-05-30:
@smhaller I added a Pull request to Moonraker: https://github.com/Arksine/moonraker/pull/436

lucaschoeneberg, 2022-05-30:
Yes, that is correct, the idea is that a dropdown menu specifies the authentication source. The preselected authentication source can be set via the default_source.
Do you think it would be useful to create another endpoint for this in Moonraker, or can we include this somewhere useful?

Extra:
Currently, I'm adding a little something so that the /access/user(get) and /access/users/list(get) endpoints have the source with them. Then we can include the source under settings->authentication.

lucaschoeneberg, 2022-06-08:
So the pull request is merged into the master. Now we only have to adjust fluidd. The functionality is currently not limited unless you change the default_source to ldap.

pedrolamas, 2022-06-08:
Thanks @lucaschoeneberg, but that commit has been temporarily reverted as some issues started to be seen while upgrading to this new version.

https://github.com/Arksine/moonraker/issues/447#issuecomment-1149738780

We will take a look into this once it gets merged back again in the future.

lucaschoeneberg, 2022-06-11:
So the feature is integrated into the master. Now some functions have to be implemented in the frontend, because if the default_source is set to LDAP, e.g. the function to add users doesn't work, and many other things have to be implemented too.

closed 2022-06-19 by matmen