fluidd-core/fluidd #421: Multiple instances of klipper/moonraker - unable to be restarted via GUI

teikjoon, 2021-09-29:
## Affected Version(s):
- klipper version: v0.9.1-786-gc2d3a168
- moonraker version: v0.7.1-56-g2447cca
- fluidd version: v1.16.2

## Platform:
Windows 10/Chrome/Ubuntu 20.04

## Hardware Configuration:
BTT SKR1.4 Turbo, Lenovo M73 Tiny

## Reproduction Steps:
1. Tried to restart klipper or moonraker service

[image]

## Actual Results:
Fluidd reports error 'unable to restart Klipper'

## Expected Results:
Klipper/moonraker should restart

## Additional information
Have installed multiple instances of klipper/moonraker using KIAUH, so the klipper service is actually 'klipper-1' & 'klipper-2' not 'klipper'. 

This is probably why it can't restart any of the services - also logged an issue with KIAUH dev and this was his response : https://github.com/th33xitus/kiauh/issues/37#issuecomment-928374965

randellhodges, 2022-02-01:
In my opinion, if I'm connected to Printer-1, I wouldn't want to restart a klipper or moonraker instance for Printer-2. Things like checking if it is currently printing, etc wouldn't really be possible.

It would be nice if Moonraker would filter or return the instance of the moonraker you are currently connected to, as well as the service name of the klipper instance that moonraker instance is connect to.

Then you'd still just have klipper/moonraker listed in the services and it would be the one that you currently have a websocket connect to.

It would work as a facade, translating klipper or moonraker to klipper-X or moonraker-X where -X is the instance.

randellhodges, 2022-02-02:
You can ignore about 1/2 of what I previously said.

I'm been playing with some service enhancements. It seems this issue is specific to the klipper instances? Moonraker will always restart the instance you're connect to?

So, a couple ways to handle this.

1) make the name of the klipper service configurable, default to klipper.
2) get a change made to moonraker so that it has an endpoint like `/server/restart` but instead of stopping the web server service, the 3 services (start/stop/restart) would signal klipper to do the actions?

I think option 2 might not be well received.

pedrolamas, 2022-03-30:
> It would be nice if Moonraker would filter or return the instance of the moonraker you are currently connected to, as well as the service name of the klipper instance that moonraker instance is connect to.

That would be nice, but I don't think it is actually possible, as there are multiple ways that it could be running and auto-discovery is not an easy thing to achieve in this case.

I think the changes introduced by @randellhodges in #504 improved this quite a lot, not sure what more we can improve here?

matmen, 2022-04-12:
Hey @teikjoon, can you confirm if #504 solved this issue? Thanks.

teikjoon, 2022-04-12:
Hello @matmen , yes #504 solved this issue. I'm now able to restart the services individually. Thanks!

[image]

matmen, 2022-04-12:
Amazing, thanks for confirming!

closed 2022-04-12 by matmen