fluidd-core/fluidd #422: FR: Pressure advance values stored with thermal presets

aepkolol, 2021-09-30:
## Is your feature request related to a problem? Please describe.
No

## Describe the solution you'd like:
Right now we currently have "thermal presets" for different materials. Can we change this to material presets, and also store pressure advance (and maybe also flow percentage) along with this? That way if we set it to ABS over PLA, it will adjust the pre-determined PA values as well?

## Describe alternatives you've considered:
Manually adjusting the config or console command

## Affected users and/or setups:
Anyone who regularly uses klippers pressure advance feature.

## Additional Context:
N/A

Afterschnitzel, 2021-10-01:
I use pressure advance preset in slicer. In every filament profile I have custom gcode for pressure advance preset, flow, firmware retraction. It is simple.
Only usable thing in Fluidd should be pressure advance horizontal scrollbar like the other settings have.

Biorn1950, 2022-01-29:
Since you need to adjust pressure advance for each filament you should profiles in your slicer. This feature could be added in a spool-manager eventually.

IMO thermal preset should be used only if you want to do things on your extruder and need it to be hot. Not for printing purpose.

closed 2022-01-29 by Biorn1950