fluidd-core/fluidd #428: Implement a button to download your logs from the side panel

KetamineKyle, 2021-10-12:
Please add buttons to the side panel that allow you to download your klippy/moonraker logs, simple as that
[image]

## Describe the solution you'd like:
A button that downloads the logs from klippy/moonraker to your device

## Describe alternatives you've considered:
SSH'ing into the machine which is a pain.

## Affected users and/or setups:
Anyone who is trying to troubleshoot their problems will benefit from this.

Biorn1950, 2022-01-29:
Are you want to be able to upload a custom logo from your computer?

Biorn1950, 2022-01-29:
You already have it here.
[image]

closed 2022-01-29 by Biorn1950