fluidd-core/fluidd #445: Pi looses WiFi connection randomly

rclarsfull, 2021-11-07:
Hey,
i dont know that log to send youz to help finde the issue.

The pi is loosing its wifi conection about ever 3-40 mins for about 1-5 mins.

I already checkt if wifi powermanegment is on, but its off.
I´ve already written a script that pings Google every minute, still the same issue.
i´ve tryd to update everything, the error still exists.

The printer remains printing fine, i just cant ping it anymore and, my rooter dosent finds him connectet and moonracker gives an connection errror.

My platform is a pi 4.

grigi, 2021-11-09:
I had a similar issue with my RPi3B. looking at `dmesg` output the wifi chip crashed continuously. Reinstalled the OS, but problem persisted. disabled built in wifi and used a USB dongle, and now everything works reliably.

There is a chance that your Pi's wifi spontaneously broke.

ChipCE, 2021-11-17:
I have the same problem. I'm not sure but tt is seem like the camera process cause other process to crash on mine. The camera frame rate will drop first and after a few minutes wifi is crashed. Disable camera on mine and it works fine since then.

closed 2022-01-29 by Biorn1950