fluidd-core/fluidd #446: Interface unstable when internet is not working

RealDeuce, 2021-11-10:
When the internet is not working (specifically, when api.google.com does not resolve), the Fluidd interface reconnects to moonraker every 5-15 seconds with the following error in the moonraker log:
```
2021-11-10 08:17:14,299 [proc_stats.py:_watchdog_callback()] - EVENT LOOP BLOCKED: 11.04 seconds
2021-11-10 08:17:14,315 [update_manager.py:github_api_request()] - Error Processing GitHub API request: https://api.github.com/rate_limit
Traceback (most recent call last):
  File "/home/admin/moonraker/moonraker/components/update_manager/update_manager.py", line 596, in github_api_request
    resp = await asyncio.wait_for(fut, 10.)
  File "/usr/lib/python3.7/asyncio/tasks.py", line 416, in wait_for
    return fut.result()
tornado.curl_httpclient.CurlError: HTTP 599: Resolving timed out after 5002 milliseconds
2021-11-10 08:17:14,501 [proc_stats.py:log_last_stats()] - 
System Time: 1636550221.147640, Usage: 5.01%, Memory: 24984 kB
System Time: 1636550222.147710, Usage: 5.85%, Memory: 24984 kB
System Time: 1636550223.149318, Usage: 5.05%, Memory: 24984 kB
System Time: 1636550224.148275, Usage: 5.0%, Memory: 24984 kB
System Time: 1636550234.315949, Usage: 0.69%, Memory: 24992 kB
```

If an entry for api.github.com pointing to localhost is added to /etc/hosts to prevent a resolution failure from occurring, the interface remains stable, despite connections failing:
```
2021-11-10 08:31:19,521 [update_manager.py:github_api_request()] - Error Processing GitHub API request: https://api.github.com/rate_limit
Traceback (most recent call last):
  File "/home/admin/moonraker/moonraker/components/update_manager/update_manager.py", line 596, in github_api_request
    resp = await asyncio.wait_for(fut, 10.)
  File "/usr/lib/python3.7/asyncio/tasks.py", line 416, in wait_for
    return fut.result()
tornado.curl_httpclient.CurlError: HTTP 599: Failed to connect to api.github.com port 443: Connection refused
2021-11-10 08:31:20,528 [update_manager.py:github_api_request()] - Error Processing GitHub API request: https://api.github.com/rate_limit
Traceback (most recent call last):
  File "/home/admin/moonraker/moonraker/components/update_manager/update_manager.py", line 596, in github_api_request
    resp = await asyncio.wait_for(fut, 10.)
  File "/usr/lib/python3.7/asyncio/tasks.py", line 416, in wait_for
    return fut.result()
tornado.curl_httpclient.CurlError: HTTP 599: Failed to connect to api.github.com port 443: Connection refused
2021-11-10 08:31:21,533 [update_manager.py:github_api_request()] - Error Processing GitHub API request: https://api.github.com/rate_limit
Traceback (most recent call last):
  File "/home/admin/moonraker/moonraker/components/update_manager/update_manager.py", line 596, in github_api_request
    resp = await asyncio.wait_for(fut, 10.)
  File "/usr/lib/python3.7/asyncio/tasks.py", line 416, in wait_for
    return fut.result()
tornado.curl_httpclient.CurlError: HTTP 599: Failed to connect to api.github.com port 443: Connection refused
2021-11-10 08:31:22,540 [update_manager.py:github_api_request()] - Error Processing GitHub API request: https://api.github.com/rate_limit
Traceback (most recent call last):
  File "/home/admin/moonraker/moonraker/components/update_manager/update_manager.py", line 596, in github_api_request
    resp = await asyncio.wait_for(fut, 10.)
  File "/usr/lib/python3.7/asyncio/tasks.py", line 416, in wait_for
    return fut.result()
tornado.curl_httpclient.CurlError: HTTP 599: Failed to connect to api.github.com port 443: Connection refused
2021-11-10 08:31:23,545 [update_manager.py:init_api_rate_limit()] - Error Initializing GitHub API Rate Limit
Traceback (most recent call last):
  File "/home/admin/moonraker/moonraker/components/update_manager/update_manager.py", line 518, in init_api_rate_limit
    resp = await self.github_api_request(url, is_init=True)
  File "/home/admin/moonraker/moonraker/components/update_manager/update_manager.py", line 641, in github_api_request
    f"Retries exceeded for GitHub API request: {url}")
utils.ServerError: Retries exceeded for GitHub API request: https://api.github.com/rate_limit
```

Adding api.github.com to /etc/hosts of course is not a long-term solution since this prevents cersion checking and auto-updates from working.

## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: v0.10.0-116-ga0615e5e
- moonraker version: v0.7.1-115-g7ff126a
- fluidd version:v1.16.2-0fa3c9b 

## Platform:
FluiddOS

## Hardware Configuration:
Raspberry Pi 3A+

## Reproduction Steps:
Internet went down, api.github.com did not resolve.  Can be reproduced by disabling DNS lookups in the OS.

## Actual Results:
Every 5-15 seconds, Fluidd would reconnect to moonraker, blocking the interface while it connected

## Expected Results:
When the internet is down, Fluidd should not reconnect to moonraker every 5 seconds.

## Log Files:

(The log files fluidd provides at Configure > System Control.)
- klipper log: [klippy.log](https://github.com/fluidd-core/fluidd/files/7514996/klippy.log)
- moonraker log: [moonraker.log](https://github.com/fluidd-core/fluidd/files/7514995/moonraker.log)

hn0550, 2021-12-10:
This problem can be solved if the software allows you to choose between automatic or manual updates

RealDeuce, 2021-12-16:
> This problem can be solved if the software allows you to choose between automatic or manual updates

I'm not sure it stays running long enough to switch to manual updates if the internet is down.

Biorn1950, 2022-01-29:
For moonraker related issue, you can use the [Klipper discord](https://discord.gg/jnRRNRn3) or the [Klipper discourse](https://klipper.discourse.group/)
FYI you should be able to disable it. https://moonraker.readthedocs.io/en/latest/configuration/#update_manager

closed 2022-01-29 by Biorn1950

RealDeuce, 2022-01-29:
I don't believe it's a Moonraker issue, but the way fluidd handles moonraker taking "too long" to respond (due to the host lookup timing out).

Biorn1950, 2022-01-30:
Did you tried to disable the update manager in moonraker?

RealDeuce, 2022-01-30:
Disabling moonraker would no doubt avoid the problem, as would powering off my printer... but having to manually do that every time my DNS service is interrupted, then remember to turn it back on when DNS is working again is not a reasonable solution to the problem.