fluidd-core/fluidd #449: progress is set to 0%

shj8893, 2021-11-21:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: v0.10.0-150
- moonraker version: v0.7.1-154
- fluidd version: v1.16.2

## Platform:
chrome, android

## Hardware Configuration:
(Information about your hardware configuration. Like MCU, host and setup)
Kingroon, 

## Reproduction Steps:
1. When I did clean install v1.16.2 to Raspberry Pi 3 and 4, and Printing someting, progress value and graph are set to 0%   
    doesn't increase during printing.
2. If I do clean install v1.15.x and update fluidd to v1.16.2, progress value and graph works well.  

## Actual Results:
(What happens after the reproduction steps have been followed.)
Progress value in the chrome screen doesn't work. Fixed to 0%
And It suddenly changed to 100% when finished printing.

## Screenshot(s):
(Images or gifs showing the problem, perhaps before/after.) 
[image]

## Expected Results:
(What *should* happen after the reproduction steps have been followed.)
Progress value must show current percentage 

## Log Files:
(The log files fluidd provides at Configure > System Control.)
- klipper log:
[klippy.log](https://github.com/fluidd-core/fluidd/files/7575820/klippy.log)
- moonraker log:
[moonraker.log](https://github.com/fluidd-core/fluidd/files/7575821/moonraker.log)

## Additional information
(Extra information relevant to the issue, links to similar issues.)

shj8893, 2021-11-24:
Today this Issue is cleared from following versions.
fluidd: v1.16.2
klipper: v0.10.0-164
moonraker: v0.7.1-169

shj8893, 2021-11-24:
Not Closed. Today again! Progress bar fixed to 0%

Biorn1950, 2022-01-29:
is it solved?

matmen, 2022-02-25:
@shj8893 Any updates on this? I assume it's still happening?

matmen, 2022-03-03:
Closing because of staleness. If this is still a problem, feel free to re-open.

Arakon, 2022-03-06:
I'm on a fresh install (using kiauh) with all the latest versions as of today and I'm seeing the same. Both in Fluidd itself, and on the display, it stays at 0% until done, then jumps to 100%.
[image]

matmen, 2022-03-06:
Just to make sure: do you have an empty `[display_status]` in your printer.cfg (as described [here](https://docs.fluidd.xyz/configuration/initial_setup#display_status))? Fluidd gets the progress data directly from Klippy, so this needs to be configured.

Arakon, 2022-03-06:
Both are set up in printer.cfg, yes.

pedrolamas, 2022-03-06:
@Arakon hope you don't mind, I have a few questions on this for us to try and figure out what is going on here...

- what browser and operating system are you using?
- any error on the browser console? (press F12 to open the Developer Tools, change to "Console" tab and refresh the page)
- what language/regional setting are you using on the Pi? and on the browser?
- can you share the gcode you used?

If you use Discord, please come by our server where we can discuss this directly! https://discord.gg/GZ3D5tqfcF

Arakon, 2022-03-06:
- Windows 10, Firefox. However, that shouldn't influence the LCD display, right?
- Language/region settings are default, US english I suppose. 
- Gcode shouldn't matter, as it happens with all of them.. still, attached.
Error log:
[image]
[3DBenchy_44m_0.2mm_205C_PLA_ENDER3.gcode.zip](https://github.com/fluidd-core/fluidd/files/8193622/3DBenchy_44m_0.2mm_205C_PLA_ENDER3.gcode.zip)

matmen, 2022-03-06:
> However, that shouldn't influence the LCD display, right?
That's right! The display is controlled by Klippy, which is also feeding the wrong data to Moonraker (where Fluidd is getting it from).
> Gcode shouldn't matter, as it happens with all of them.. still, attached.
If it happens on all prints, this makes me believe that Klippy isn't actually parsing the status display config section. Can you upload your printer config file(s) too, please?

pedrolamas, 2022-03-06:
> * Gcode shouldn't matter, as it happens with all of them.. still, attached

Just tested that gcode file you sent on my simulavr and can reproduce the issue, so definitely something there! 🙂

Will dig into it tomorrow and report back my findings!

Arakon, 2022-03-06:
> That's right! The display is controlled by Klippy, which is also feeding the wrong data to Moonraker (where Fluidd is getting it from).
>
>
> If it happens on all prints, this makes me believe that [...]
[printer.zip](https://github.com/fluidd-core/fluidd/files/8193683/printer.zip)

matmen, 2022-03-07:
Looks like SuperSlicer is emitting wrong "Print remaining times"-gcodes to your file:
```gcode
M73 P0 R43
M73 Q0 S43
```
The first `M73` looks fine, but it reset by the 2nd one (`M73` doesn't support a `Q` parameter). Can you try re-slicing with "G-code flavor" set to "Klipper" under "Printer Settings" in SuperSlicer?

Arakon, 2022-03-07:
Ahhh, sorry! I wasn't even aware that there was a Klipper option for the gcode flavor. Last time I had modified anything there, there was reprap and marlin and I was still using slic3r.
Will try it that way later.

pedrolamas, 2022-03-07:
@Arakon Using Marlin as G-code flavor is perfectly fine (though I would still recommend using Klipper).

I've found the reason for this `M73 Q` output: the "Firmware: Supports stealth mode"

[image]

If that option is disabled, it should work correctly.

pedrolamas, 2022-03-07:
According to [this](https://marlinfw.org/docs/gcode/M073.html), **Marlin does not support** `M73 Q`, so I would consider this a Slicer issue.

pedrolamas, 2022-03-07:
I've raised bugs for this in both PrusaSlicer and SuperSlicer:

- https://github.com/supermerill/SuperSlicer/issues/2483
- https://github.com/prusa3d/PrusaSlicer/issues/8036

Arakon, 2022-03-08:
That did indeed fix it.

pedrolamas, 2022-03-08:
Good news: I've pushed a PR to fix this upstream in Klipper, and that has been accepted and merged, so I'm now closing this issue!

closed 2022-03-08 by pedrolamas