fluidd-core/fluidd #450: Thermals Chart is empty

Greendumb, 2021-11-22:
## Affected Version(s):
- klipper version: v0.10.0-153-g4eeb4620
- moonraker version: v0.7.1-165-g97e2766
- fluidd version: v0.7.1-165-g97e2766

## Platform:
The problem occurs on all platforms I use: desktop (win 10 all browsers), smartphone (iOS 15) etc

## Hardware Configuration:
Raspberry Pi 4 Model B

## Reproduction Steps:
Installed on a clean distribution Raspberry Pi OS via KIAUH
Tried to install twice from scratch, but it did not work

## Actual Results:
There is no result, Thermals Chart is also empty

## Screenshot(s):
[image]

## Log Files:
- klipper log: [klippy (8).log](https://github.com/fluidd-core/fluidd/files/7578986/klippy.8.log)
- moonraker log: [moonraker (1).log](https://github.com/fluidd-core/fluidd/files/7578989/moonraker.1.log)

## Additional information
(Extra information relevant to the issue, links to similar issues.)

Afterschnitzel, 2021-11-22:
Same here. It was caused by updating moonraker.

DonyBerserker, 2021-11-23:
Same here, after update moonraker

Greendumb, 2021-11-23:
The latest moonraker update (v0.7.1-169-g50fd8c9) solved the problem

closed 2021-11-23 by Greendumb

hieppo, 2021-11-25:
Unfortunately, v0.7.1-169-g50fd8c9 does not solve the issue. I have updated to currently v0.7.1-173-gccb86ea and the thermal chart are not displaying.