fluidd-core/fluidd #463: Cannot use Chinese as gcode file name

clementatt, 2021-12-30:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version:v0.10.0-215-g9bdd6175
- moonraker version:v0.7.1-280-g9d7baa1
- fluidd version:v1.16.2

## Platform:
I use Windows 11 21H2 and Chrome 96.0.4664.110
## Hardware Configuration:
I use Raspberry Pi 4B 2G as the host computer and mks gen_l v2.1 （mega2560）as the main board of the 3D printer.

## Reproduction Steps:
after I upgraded to v1.16.2 version, there was a problem that Chinese could not be used as the gcode file name.
## Actual Results:
(What happens after the reproduction steps have been followed.)

## Screenshot(s):
[image]
[image]
[image]

## Expected Results:
(What *should* happen after the reproduction steps have been followed.)

## Log Files:
(The log files fluidd provides at Configure > System Control.)
- klipper log:
[klippy.log](https://github.com/fluidd-core/fluidd/files/7793339/klippy.log)

- moonraker log:
[moonraker.log](https://github.com/fluidd-core/fluidd/files/7793340/moonraker.log)

## Additional information
(Extra information relevant to the issue, links to similar issues.)

sjqlwy, 2022-01-06:
https://mc.dfrobot.com.cn/thread-311545-1-1.html?fromuid=725344

BastelKlug, 2022-01-08:
I think the problem is with klipper or moonraker and not with fluidd.

Biorn1950, 2022-01-29:
Hello,

For Klipper related issue, you can use the [Klipper discord](https://discord.gg/ctSg5BpB) or the [Klipper discourse](https://klipper.discourse.group/)

- Biorn

closed 2022-01-29 by Biorn1950