fluidd-core/fluidd #464: Multiple printers setup not available from a different browser

gsdaemon, 2022-01-02:
Hello

It seems that on a multi-printers setup, the configured printers are saved now only inside localStorage of the browser and, when opening the Fluidd interface from another browser / device, it will only see the default printer which is returned by the apiserver located at the default moonraker address:port (127.0.0.1:7125). (Or maybe I'm doing something wrong)
[image]

Would it be possible to save the multi-printers config independently and have them available across multiple clients? Or is there maybe a workaround?

Thanks

shinuza, 2022-01-30:
The configuration is per device by design, this allows us to have a different layout on mobile and desktop for example. We could look into it but this might take some thinking to figure out which settings should be in global (moonraker) conf and which should be saved locally.

gsdaemon, 2022-01-30:
Thank you for your reply.
The "per browser" configuration definitely makes sense when it's UI related, but printer instances config imho could be considered more general/global, as they normally have nothing to do with the browser you're opening Fluidd from.

lucaschoeneberg, 2022-02-03:
Hi, we also use Fluidd for our 3D printers so that our students can observe them from the outside. The current problem is that we always have to give the students the IP addresses of the printer so that the students can get the printer in their browser. So it would be helpful if there were a setting in the fluidd config for the default printer. It is currently written to LocalStorage under: appInstances.

shinuza, 2022-02-03:
I agree this would be a good addition, but implementing this without an opt-in/opt-out would "break" standalone installation like https://app.fluidd.xyz

Skiddz, 2022-02-20:
I'm seeing the same issue as the OP, but my additional printers are not "sticking" between Fluidd sessions.  I can add printers and interact with them, but if I shut down the browser (or even just the tabe) and open a new session later, only the default printer is listed and I have to add it again.

closed 2023-01-29 by matmen