fluidd-core/fluidd #465: ESP32 cam as wifi ip cam for fluidd??

tnap1979, 2022-01-02:
Hey,
I have a ESP32 cam, a few of em..
And I'd like to be able to use them as wifi ip cams in fluidd... is this even possible??
All I have is an ip address and it streams thru an interface
[image]

BastelKlug, 2022-01-08:
Have you tried this? https://docs.fluidd.xyz/features/cameras

tnap1979, 2022-01-08:
I have, and all I got was no display when I entered my ip address of the esp32 cam in there

Biorn1950, 2022-01-29:
Hello,
For webcam related issue you can use the [Klipper discord](https://discord.gg/eEZr8Qvc) or the [Fluidd discord](https://discord.gg/ccF72ypK)

Biorn.

closed 2022-01-29 by Biorn1950