fluidd-core/fluidd #466: Fluidd v1.16.2 deleting jobs when selecting all but one object

stevereno30, 2022-01-04:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: v0.10.0-185-gdebcc22f
- moonraker version: v0.7.1-253-g3e70e4b 
- fluidd version: v1.16.2 

## Platform:
Firefox, Windows 11

## Hardware Configuration:
RaspPi 3B, BTT Octopus

## Reproduction Steps:
1. Select jobs/folders with the checkboxes until only one remains unchecked.

## Actual Results:
The final box is automatically selected.

## Screenshot(s):
[image]

## Expected Results:
The final box should remain unchecked.

## Log Files:
(The log files fluidd provides at Configure > System Control.)
- klipper log: [klippy.log](https://github.com/fluidd-core/fluidd/files/7810617/klippy.log)

- moonraker log: [moonraker.log](https://github.com/fluidd-core/fluidd/files/7810619/moonraker.log)

BastelKlug, 2022-01-09:
Hello, I have tried to reproduce your bug with the newest fluidd dev version (1.7.0). Maybe it was fixed. 

Sorry for the bad quality:
[image]

BastelKlug, 2022-01-09:
I have found the PR which fixes this Bug: https://github.com/fluidd-core/fluidd/pull/443

closed 2022-01-29 by Biorn1950