fluidd-core/fluidd #468: CANCEL_PRINT not accepted in Fluidd.cfg

pgale, 2022-01-09:
## Affected Version(s):
(The versions of fluidd this issue occurs on.)
- klipper version: v0.10.0-223-g4c8d24ae
- moonraker version: v0.7.1-285-g505c1bf
- fluidd version: v1.16.2

## Platform:
(Information about your platform, like the OS or browser you are using.)
Raspberry Pi 4

## Hardware Configuration:
(Information about your hardware configuration. Like MCU, host and setup)
Ender 3 v2 4.2.2

I keep getting the 'CANCEL_PRINT macro not found in configuration.' error if I remove the CANCEL_PRINT macro from printer.cfg and add it to Fluidd.cfg

Docs seem to suggest (but not clear) that Fluidd.cfg holds this macro. Does some version also need to be in the Printer.cfg and if so, who takes precedence?

BastelKlug, 2022-01-09:
did you include the Fluidd.cfg with [include Fluidd.cfg]?

pgale, 2022-01-09:
Should that go in printer.cfg?
I’ve not done that if so

BastelKlug, 2022-01-10:
Yes you have to do that because with that external file you are expand the printer.cfg

pgale, 2022-01-12:
Thanks - will try that.

closed 2022-01-12 by pgale