fluidd-core/fluidd #494: Force Move switch

Biorn1950, 2022-01-29:
### Is your feature request related to a problem? Please describe

a

### Describe the solution you'd like

a

### Describe alternatives you've considered

a

### Additional information

a

pedrolamas, 2022-03-27:
This sounds interesting and I think we should take a look into it, but how would this really work? What commands would we be sending to Klipper?

matmen, 2022-03-27:
> This sounds interesting and I think we should take a look into it, but how would this really work? What commands would we be sending to Klipper?

FORCE_MOVE is the command you're looking for 😉

pedrolamas, 2022-03-27:
Given this most likely will put the printer in an incorrectly homed stat, I wonder if it would make more sense to have it on the "Tune" panel instead of the main one?

matmen, 2022-03-27:
I think this would make sense if the printer isn't homed at all, with big warnings telling users that it's a bad idea and may physically break the printer. Maybe a checkbox (or a button to toggle until next home) in a menu in the tool card header that just enables the movement arrows?

PR #750 (2022-07-15): feat(ui): Add FORCE_MOVE support to tool controls
Closes: #494 

Hi,
I am here with another PR 😁 I hope I have not misunderstood the request.

I have made a video again to demonstrate the function:

https://user-images.githubusercontent.com/44060099/177890838-768aa72c-2c1f-42e2-b1f1-78fba8b4be3b.mov

I have not included a warning, anyone who manually activates force_move should actually know the dangers. Personally, I would find it annoying. If it is still desired, I can work on it.

This PR is not complete and needs some css work. I've been working on it for days but failed, maybe someone els [... cut]
closed 2022-07-15 by matmen