fluidd-core/fluidd #501: Layer count is wrong when printing multiple object with "complete individual objects" enabled in Super Slicer

RobRobM, 2022-01-31:
### What happened

I have printed multiple projects from Super Slicer today with 2 or more object completed individually and the layer count is wrong for all but one object.
[image]
.

### What did you expect to happen

See the layer count of each individual object or the total number of layers of all object combined.

### How to reproduce

Slice multiple objects in Super Slicer with "complete individual objects" enabled, print and watch the layer count.

Biorn1950, 2022-02-02:
Hello, I think the total combination is the way to go.

RobRobM, 2022-02-02:
I second that.

pedrolamas, 2022-03-06:
Arguably, we might want to look into not only showing the current and total layers, but also the current and total number of objects to print!

pedrolamas, 2022-03-11:
@RobRobM can you provide the gcode so we can repro this issue?

PR #572 (2022-03-12): fix: individual object layer count
Fixes #501
closed 2022-03-12 by pedrolamas