fluidd-core/fluidd #502: unable to reboot pi, shut down pi after updates on fluidd/moonraker/klipperscreen

buikpijn, 2022-02-01:
### What happened

After updating through the Fluidd webinterface klipperscreen, fluidd and klipper i am no longer able to perform the remaining update of the system ('error updating system') sudo apt update/upgrade perform fine through putty. But the update button keeps showing up for system. 

### What did you expect to happen

I expect it to update the system through the Update button

### How to reproduce

Click the system restart or system shutdown button.

### Additional information

The Host>shutdown  feature for the Pi also doesn't work from the webinterface of Fluid and sudo reboot/shutdown work just fine through putty.  Available update under the system update are 

libpolkit-agent-1-0
libpolkit-backend-1-0
libpolkit-gobject-1-0
policykit-1
rpi-eeprom

buikpijn, 2022-02-01:
I think i found something;

'Moonraker not authorized for PolicyKit action'

'Moonraker not authorized for PolicyKit action: [org.freedesktop.login1.reboot], The reboot API will be disabled'

How to correct this?

buikpijn, 2022-02-01:
Reboot part solved. Per manual, it is NOT ./moonraker/scripts/sudo-fix.sh but ./moonraker/scripts/sudo_fix.sh 

And that didn't solve it either, but ./set-policykit-rules.sh resolved it. I am starting to hate it more and more. What a huge waste of time.

Arksine, 2022-02-02:
FWIW, `sudo_fix.sh` can now be considered deprecated.  That is one of the main goals of the changes that required policykit.  For more information, see https://github.com/Arksine/moonraker/issues/349

Titaniser, 2022-02-04:
i had the same problem but with mainsailos.
ity a moonraker problem
i found a fix on [https://moonraker.readthedocs.io/en/latest/installation/#policykit-permissions](https://moonraker.readthedocs.io/en/latest/installation/#policykit-permissions)

buikpijn, 2022-02-04:
To be honest, I discovered it was Moonraker after i installed mainsailos. (this OS gives a clear error on the first page of the control of the printer, unlike Fluidd).

In my experience every update involves the risk of loosing a working printer. All the tools need to coorporate perfectly, someone updates something somewhere in their policies or uses a different python version and you're screwed after simple updating all the items required to run your printer (in my case, klipper, Fluidd, moonraker, klipperscreen and the main distribution of the OS. It's not the first time that my printer broke down after an update. I admire all the hard work that has been put into this to make it work all together. I have this experience with everything running on a Pi /linux. It would be fair easy to re-install everything, but with an adxl345 with the required dependencies in the mix and a MCU that doesn't automatically updates (need to be done by hand through the SD card) it is easier said than done.