fluidd-core/fluidd #516: Idle time does not start counting after a firmware restart.

RobRobM, 2022-02-08:
### What happened

The idle time does not start counting after a firmware restart. Only after a move, setting bed or extruder to heat or a print does the timer start counting. 

In my case this is the code that works, only after a printer move, heating or print:
[idle_timeout]
gcode:
  M84
  TURN_OFF_HEATERS
  UPDATE_DELAYED_GCODE ID=delayed_printer_off DURATION=600

Not the biggest deal, because I this now, and it works after a print is done. But it would be nice if it always starts the countdown.

### What did you expect to happen

(Re-)Starting the idle countdown on firmware restart.

### How to reproduce

Set idle timeout and an action after the timout to do something. Restart firmware and after the timeout has counted down nothing happens.

randellhodges, 2022-02-08:
This would be a klipper bug, not a fluidd issue

RobRobM, 2022-02-08:
Thanks for the info, I'll move it there.

closed 2022-02-08 by RobRobM