fluidd-core/fluidd #518: Latest update: Host/Reboot/Shutdown no longer work

imacmill, 2022-02-10:
### What happened

fluidd
v1.16.2
klipper
v0.10.0-256-g6d7c0336
moonraker
v0.7.1-356-g9e9f9ee

Neither the Reboot nor Shutdown options in the Host menu work after the latest update. Running on rPi 4.

### What did you expect to happen

Clicking wither Shutdown or Reboot displays a confirmation prompt, and after clicking 'Yes', nothing happens...the system stays up.

### How to reproduce

Clicking wither Shutdown or Reboot displays a confirmation prompt, and after clicking 'Yes', nothing happens...the system stays up.

yzheka, 2022-02-11:
It is moonraker issue. To fix it run set-policykit-rules.sh which is located in ~/moonraker/scripts directory

imacmill, 2022-02-13:
I ran the indicated script as follows:

sudo sh set-policy-kit-riles.sh

The following was printed to the terminal:

***************************

set-policykit-rules.sh: 108: [: Illegal number:
-e

###### PolicyKit Version 0.105 Detected
-e

###### Installing Moonraker PolicyKit Rules (Legacy) to /etc/polkit-1/localauthority/50-local.d/10-moonraker.pkla...
-e

###### Restarting Moonraker...
****************************

I ignored the seeming error, kept ,y PuTTY session open, loaded FLUIDD in my browser, and got the following red error dialog when I load FLUIDD in my browser

"Metadata not available for"   [CLOSE]

The 'Host/Shutdown&&Reboot' menu items still do not work.

I disconnected PuTTY session, reloaded FLUIDD in browser, error dialog gone but shutdown/reboot menu items still do not work.

yzheka, 2022-02-13:
Hmm. That was helped for me. Maybe try reinstall script for moonraker?

yasaslive, 2022-02-15:
@yzheka I had the same issue! It works! Thanks

pedrolamas, 2022-02-16:
Running the following commands should fix the issue:

```sh
cd ~/moonraker/scripts  
./set-policykit-rules.sh  
sudo service moonraker restart
```

If this doesn't fix it for you, I assume that there is some other difference in your system (what device are you using? what OS/distro?)

imacmill, 2022-02-17:
As per my first post, I'm running this on a Raspberry Pi 4, and I ran the script you posted, which generated an error during execution. I did not issue a service restart, but instead completely rebooted the Pi. The Host command still do not work.

I installed Fluidd two months ago using the lates Fluidd image available at the time (and that I have long since deleted, so I don't know what version it was). It was an update installed thru the browser UI a few days ago that started the trouble.

pedrolamas, 2022-02-17:
What user are you using to run those commands? `pi`, `root` or other?

imacmill, 2022-02-17:
su.

pedrolamas, 2022-02-17:
Can you try running as `pi` instead?

imacmill, 2022-02-18:
Not sure why, but my replies via email have not been working. I emailed yesterday that Pedro's suggestion worked.

All is well here...thanks.

pedrolamas, 2022-02-18:
Excellent, closing this now then! 🙂

closed 2022-02-18 by pedrolamas