fluidd-core/fluidd #519: Gcode live view not working for circular bed (delta printer)

yzheka, 2022-02-11:
### What happened

Gcode live view is empty when using on detla printer but it is working good on rectangular bed. I guess it is because of round bed have origin at 0;0
[image]

### What did you expect to happen

Gcode live view is visible when using round bed

### How to reproduce

Start any print on delta printer and load current file to gcode live view

Mechazawa, 2022-02-22:
I sadly don't own a delta printer nor do I have a slicer set up for one. Could you share some gcode for testing?

yzheka, 2022-02-22:
@Mechazawa
[test_cube.zip](https://github.com/fluidd-core/fluidd/files/8119570/test_cube.zip)
Here. I have sliced test cube with cura using my regular settings

PR #591 (2022-03-19): feat: adds delta support to gcode viewer
Fixes #519
closed 2022-04-26 by pedrolamas