fluidd-core/fluidd #520: Audio Cues for printer paused and finished in browser

mrfloppy88, 2022-02-13:
### Is your feature request related to a problem? Please describe

when printer goes in pause because of fillament runout or a M600 and printer is not near the user watching on browser
it can go by unnoticed and ruin some prints

### Describe the solution you'd like

upload sound files or pick between built in sounds to play when printer is done or paused

pedrolamas, 2022-02-18:
Indeed, this can be added to Fluidd or any other Klipper interface, but I do wonder if using the [native support in Klipper](https://github.com/Klipper3d/klipper/blob/8b0c6fcb089769f70ecbb11cc3793dcd61f445dd/config/sample-macros.cfg#L64-L86) isn't a preferred method (as that will work even if no graphical interface is connected!)

In the case of Fluidd, for this to work it would mean that you have a computer with the browser pointing to Fluidd and that the audio is not mute... 🤔

Biorn1950, 2022-02-22:
I'm agree with @pedrolamas we must use the native notification.

matmen, 2022-02-25:
Related: #129

kthornbloom, 2022-03-07:
I started toying with this by adding

```
/* eslint-disable */
  @Watch('estimates.progress', { immediate: true }) onMatchChanged () {
      var progressPercent = parseInt(this.estimates.progress)
      if(progressPercent === 100){
        const greeting = new Notification ('FLUIDD', { body: 'Operation Completed' })
      }
  }

  mounted () {
    if (Notification.permission !== 'granted') {
      Notification.requestPermission()
    }
  }
```

to statustab.vue. Works well, but needs more tweaks to know if the job just completed vs a page refresh. Unfortunately, as a vanilla JS guy, all this typescript + hardcore linting is killing my soul 😄

matmen, 2022-03-14:
Closing in favor of #129

closed 2022-03-14 by matmen