fluidd-core/fluidd #521: Option to disable pi voltage/throttle warnings

frap129, 2022-02-13:
### Is your feature request related to a problem? Please describe

I'm getting a lot of noitifcation spam because of the undervoltage notifications. My power supply is good enough, but not necessarily stable. Voltage dips a little here and there, but I've been running my printer with it for over a year and have not had any issues related to it.

### Describe the solution you'd like

A toggle in settings to shut off these notifications would be nice.

frap129, 2022-02-16:
@pedrolamas Nope, already tried that and it does not hide the warnings. As far as I can tell, that config has been deprecated.

pedrolamas, 2022-02-16:
Yep, I remember now, it has been superseded by this: https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#using-dvfs

frap129, 2022-02-16:
That only seems to apply for the raspberry pi 4, I'm using a raspberry pi 3b+ (guess I should have mentioned that in the issue). I'm not sure if the config actually removes the on-screen warning overlays as I only use my pi over SSH and through fluidd, but the "avoid_warnings" does not stop the notifications in fluidd.

frap129, 2022-02-17:
I've worked around this issue by renaming `/opt/vc/bin/vcgencmd` to `/opt/vc/bin/vcgencmd.backup` and replaced the original binary with an empty shell script that only has a shebang at the top. I'll leave this open because I still think a simple toggle in fluidd would be a better solution, but I don't have any experience with Vue and very little with js/ts, so this "solution" is good enough for me

matmen, 2022-02-19:
@frap129 Is it just the `Under-Voltage Detected` warnings you want to have an option for, or also all the others?

frap129, 2022-02-20:
> Personally, I think there's a good reason for these warnings and ignoring them is of not a fix. However, I do support this suggestion, because in the end it's still the end users decision.
I agree, I'm probably one of the few that would actually rather disable it than getting a slightly better power supply. It might be a good idea to add a warning modal to try to explain the value of the notifications to the user? Again, I don't have much experience in web dev so I'm not sure how much more effort that would be. 
 
> Is it just the `Under-Voltage Detected` warnings you want to have an option for, or also all the others?
I only get the `Under-Voltage Detected` warning occasionally, the main one that's been bothering me is `Currently Throttled`

matmen, 2022-02-20:
I've discussed this with the team, and we've come to the conclusion that this is out of scope for Fluidd.
The warnings have a good reason, and letting people simply turn them off may result in more issues than it solves (i.e. people complaining about failed prints when they don't even remember turning these notifications off).
If people really want to turn them off and know what they're doing, your workaround is the way to go.

closed 2022-02-20 by matmen

frap129, 2022-02-21:
> I've discussed this with the team, and we've come to the conclusion that this is out of scope for Fluidd. The warnings have a good reason, and letting people simply turn them off may result in more [...]

Understandable, thanks for considering it

Stooovie, 2024-05-03:
As of May 2024, replacing vcgencmd wigh just a shebang doesn't remove the Currently throttled warning.