fluidd-core/fluidd #528: Separate XY and Z homing buttons have disappeared

matthewlloyd, 2022-02-19:
### What happened

I believe [this](https://github.com/fluidd-core/fluidd/pull/478) merged PR has removed the separate XY and Z homing buttons on printers that use a homing override. I use the Klicky homing override which does have the option to home X, Y, or Z, separately, but the UI now shows only the XYZ homing button on startup:

[image]

After homing, the XY and Z homing buttons now disappear:

[image]

The homing override looks like this:

```
[homing_override]
axes:                             xyz
gcode:
   ...
        {% if 'X' in params %}
            {% set X = True %}
        {% endif %}
        # etc. for Y and Z
   ...
```

### What did you expect to happen

The XY and Z homing buttons should always be displayed.

### How to reproduce

* Run latest Fluidd develop branch.
* Define a homing override.

pedrolamas, 2022-02-20:
I can confirm that this change in behavior was introduced with that PR, but it seems that the new behavior does match the Klipper docs:

[image]

This is not something I personally use, so I'm sure I'm missing something here!

matthewlloyd, 2022-02-20:
Thanks for taking a look! I think there are two issues then:

1. There is still a variance between the new Fluidd behavior and the current homing_override docs. For example, if we have a homing_override with "axes: z", then G28 X or G28 Y or G28 XY will still run the original Klipper homing, so the UI should still be displaying the X, Y, and XY homing buttons.

2. The Klipper docs don't accurately reflect the way homing_override is being used in practice, e.g. by the commonplace Klicky probe configs. I think this is a documentation error, because it's certainly possible to have a homing_override with "axes: xyz", then run G28 XY such that only XY get homed and everything works as expected (i.e. X and Y go into the homed state, while Z is still not homed). Looking at `homing_override.py` in the Klipper source, I don't see any reason for a requirement that the gcode does actually home all three axes - it just needs to home at least all the axes specified in the parameters (which of course means it needs to be *capable* of homing all three axes). This should probably be fixed in the Klipper docs...

In general, since it is always possible to run G28 XY, G28 X, G28 Y, G28 Z, or G28 XYZ manually, I think it would be simpler and more useful to show all the buttons no matter what homing_override is set and no matter what axes actually get homed. If nothing else, it avoids the awkward blank spaces between the arrow buttons.

PR #529 (2022-02-20): fix: reverts homing override checks in ToolheadMoves.vue
This partially reverts commit d9316467e172738a855e42d08fe2db2bbff1bb50

Fixes #528
commit cc000e05 (2022-02-20): fix: reverts homing override checks in ToolheadMoves.vue (#529)

This partially reverts commit d9316467e172738a855e42d08fe2db2bbff1bb50

Fixes #528
closed 2022-02-20 by pedrolamas